bambulab/BambuStudio #1707: Wrong encoding - cyrillic

AndylgTom, 2023-04-29:
Wrong encoding is set in many places - cyrillic letters are displayed as ?????? (some of them are on screenshots).
In the initial versions of prusa slicer there was the same problem, but now it has been fixed a long time ago.
Please fix this in BambuStudio.

[image]
[image]
[image]

LiZ-Li-BBL, 2023-06-05:
How did you set the BambuStudio to cyrillic letters? I need to know the steps to reproduce this.

AndylgTom, 2023-06-07:
> How did you set the BambuStudio to cyrillic letters? I need to know the steps to reproduce this.

I just replace any language *.mo file (for example, sv, or whatever) with a Russian translation file, and then I choose this language in the program settings.

LiZ-Li-BBL, 2023-06-20:
We will handle it. can you help provide your Russian *.mo file to us?

AndylgTom, 2023-06-20:
> We will handle it. can you help provide your Russian *.mo file to us?

Okay, I just wanted to add it after you fix the encoding bug.
This is not the final translation  [BambuStudio.zip](https://github.com/bambulab/BambuStudio/files/11795942/BambuStudio.zip)

LiZ-Li-BBL, 2023-07-12:
We plan to add Russian to preferences settings in the future. You need to select the correct language from the preferences settings so that part of the text which appears as "?" will be loaded properly. 
Thank you for the translation files.

closed 2024-01-16 by LiZ-Li-BBL