bambulab/BambuStudio #1709: Make naming plates easy

julie777, 2023-04-29:
**Is your feature request related to a problem? Please describe.**

I am not even sure there is a way to name a plate.

**Describe the solution you'd like**

There are several ways to name a plate that would would. One or more of them should be implemented.
  - click on the name of the plate in the workspace and a textbox opens to allow naming.
  - show the plate names in the plate list in review (and add that list to prepare) and right click to change properties.
      - change name
      - change locked state
      - bed type
  - allow changing the plate name in the object list for the process settings

In addition, get rid of the plate number at the lower right of the plate, and show the default name instead of "Untitled". The default names already show in the process object list as "Plate 1", etc.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Haidiye00, 2023-05-04:
[image]

At present, a new disk name naming function has been added, as shown in the above figure. It will be released in future versions, Stay tuned. If you have any new needs, you can continue to raise them. Thank you

julie777, 2023-12-13:
The current implementation does make it possible to name a plate, but it is not obvious. 
It would be nice if just clicking the displayed plate name would make it editable in place. 
It would also be helpful if the plate could be renamed in the plate list in the settings section.

Haidiye01, 2023-12-14:
Please download version 1.8 to try it out. We have added an edit button, and the plate name in the ui list on the left can also be modified by clicked

closed 2024-01-16 by Haidiye00