bambulab/BambuStudio #1712: Plate locking is incomplete and allows accidental changes.

julie777, 2023-04-29:
**Describe the bug**

A locked plate should prevent change of any kind.

Currently, locking a plate does not actually lock it to prevent change. 
Locking only prevents auto-arrange operations from affecting the plate. 
  - auto-arrange ignores the locked plates
  - plate arrange is disabled
All other operations are still possible with objects on a locked plate.

Even import will place objects on a locked plate if that plate happens to be selected.

This is counter intuitive, and defeats the purpose of having the ability to lock plates.

Note: In an environment with multiple plates available, it makes more sense that any objects imported would go on an empty plate if one exists, or a new plate would be created for them. This prevents import from messing up plates already arranged. See #1711 for additional information.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Create multiple plates
2. lock one plate
3. drag an object to the locked plate
4. drag an object from the locked plate

Other examples in the description above.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

StoneLiBambulab, 2023-05-23:
it is only effective for auto arrange and auto orient now. 
the lock is designed for that do not let the automatic actions change the previous manual adjustment.

Thanks!

Stone Li

julie777, 2023-05-23:
@StoneLiBambulab 
I understand how it works currently. I am just suggesting that it could be improved. For example, if I have a plate locked and import and STL it will be placed on a locked plate if that is the selected plate. 
It is also very easy to accidentally move something on a locked plate. I just think locked should be truly locked.

StoneLiBambulab, 2023-05-29:
We'll talk about whether we need to optimize this logic.

Thanks!

Stone Li

RyanEwen, 2023-11-22:
I just noticed the Lock option for the first time and was immediately confused by it. I locked the plate, then moved something, and it allowed this.. I expected it to prevent me from changing _anything_ about the plate and items on it. Had to search to find out what it actually does.

varionix, 2024-01-14:
Locking objects on a plate is all the more important when there are a lot of objects. I have plates with up to 20-30 objects that come from a step file, i.e. mainly texts that have to be displayed in a different colour than the main object (with a depth of 0.2mm). Unintentional moving of objects which is not recognised immediately leads to fatal print results! This is unnecessary consumption of time and resources because of a fail print !! So Please ... locking a plate must lock the plate and the contents at all !!!

donhohler, 2024-11-05:
You guys really should fix this. Very aggravating.