bambulab/BambuStudio #1713: Enable "Change Filament" Option when printing in Spiral Vase mode #891

LidorBaum, 2023-04-29:
When in preview mode, after slicing a model in vase mode, in PrusaSlicer it's possible to press on the + icon in the right layers-slider and add a filament change command.
In Bambu Studio and Orca Slicer this feature is disabled. Is there an option to make it enabled?
Its working pretty good on prusa Mk3s+.
[image]
[image]

SaltWei, 2023-05-04:
> Its working pretty good on prusa Mk3s+.

This is prusa behaviour, seems not working.
[image]

LidorBaum, 2023-05-04:
@SaltWei I do not know what exactly you are using, maybe it is because of the MMU, but on normal 1-extruder 1-color prusa slicer version it is working well, here is a video
is there a command that I can add manually to change the color with the AMS? something like the M600

https://user-images.githubusercontent.com/86967825/236258302-5b9ae4f3-7479-4fd7-bf9c-68dbbddf896c.mp4

LidorBaum, 2023-05-05:
Thank you for trying and uploading the video.
I understand that "Change Filament" or "Change extruder" is something that is more complicated than just sending an M600 command? 
I wanted to use the biggest advantage - the AMS. when printing vases in vase mode, and change the color automatically

JonBerlin, 2024-06-10:
Please make this work Bambu-Team!
It shouldn't be that hard because it is possible to slice in spiral vase mode and then change the gcode file to include the color change.

LouisH98, 2024-06-23:
+1 this is really disappointing that I can do this smoother on a Prusa MK3s than I can my BBL P1S....