bambulab/BambuStudio #1715: Height range modifier interface needs work to make it more usable.

julie777, 2023-04-29:
## Describe the bug

While adding the height range modifier solves many problems and is a great feature, the current interface is not as usable as it should be.

In the object list, right click an object and select height range modifier. This creates the first range with default values. Then the usability problem start.

1.  the range is down at the bottom of the screen and far away from where the user will naturally focus. The user will be focused on the range created as part of the object. 
    [image]
2.  The first thing the user is likely to want to do is change the range for the layer.
    - the layer dividing plane is not shown on the object on the canvas that is being modified, so the user is left wondering how to change the split point for the layer If they have used Prusa Slicer (when cut only worked on height), then they are expecting a layer plane that can be dragged to change the range for the layer. 
    [image]
    - the range that was created is not selected, so there is no obvious way to change the settings for the range.
    [image]
    - Only after selecting a range belonging to the object is the user shown how they can adjust settings for the range.
    [image]
3. When the user notices the range fields at the bottom of the screen and clicks in one to change the value, only then will they see the plane for the range point. But it is going to disappoint them.
      a.  the plane is not draggable to easily move it to the desired position based on the object geometry. 
[image]
      b. in fact, clicking on the plane to attempt to drag it will remove it from the workspace 
         and collapse the object in the object list taking the user completely out of the layer editing that they were trying to do. 
      [image]
4. When the user has finished with settings for a layer and is ready to create another, they will not be able to find a way to create a new layer because the plus button to create a new range is hidden when a layer is selected
    [image]

### All of these issues combine to confuse the user and make the feature hard to use.

## Suggested Solution
1. Put the button to add a layer on the layers container line. That way is is always visible and obvious. Keep it always enabled, to allow the user to add a layer whenever they are ready.
2. allow the user to change the layer range in the layer entry instead of having a separate entry field at the bottom of the screen
3. when the user adds a range, either the initial one or a new one, select it immediately and show the plane for the top of the range.
4. make the plane draggable so the user doesn't have to calculate actual dimensions to change a range. They should be able to type in values, but they should not have to.

My crude illustration of 1 and 2
[image]

## Remaining Questions

Currently when height modifier is selected for an object two layers are created. The first layer is explicit and is 0-2mm. There is also a second implicit layer also created from 2mm to the top of the object that is not shown.
- should the implicit layer ranges be shown?
- for a given range, can the user adjust both top and bottom, for example changing the bottom of the first layer created to 1-2mm, which creates another implicit range?
- when adding the height modifier to an object should it start with 0-2 and 2-top with 0 and top not modifiable?
- should it be more obvious that a range is non-inclusive of the bottom value and inclusive of the top so that the initial range is 0< layer <= 2mm? 
- the reality is that the user selects the point at which a layer ends and there are always n+1 layers for n splits. 
  - What if the selection of the height modifier always just created two layers with 0 and the top being fixed, and when there are two adjacent layers, changing the bottom of one also changed the top of the one below it to match. 
  - The user would have to explicitly create a new layer, whereas in the current implementation they can change the initial layer starting at 0 to start at 1 and that creates another implicit layer.
- that leads to another possibility that with 2 initial layers created, the only place to create a new layer would be between them given the top and bottom of the object are fixed. So should all layers but the top one have the add layer button to add a new layer after the one with the button that was pushed?

## Desktop (please complete the following information):
 - OS:  Windows]
 - Version 1.6.0.65

zoroseerus, 2023-08-08:
Totally agree. Hope this gets seen and taken care of

SaltWei, 2024-01-19:
> Hope this gets seen and taken care of

Really sorry for seeing this much much later!

Really good suggestions, I will discuss this with developer and UI designer.