bambulab/BambuStudio #1718: Too close warning is useless and counter-productive due to constant false positives

AdrianGarside, 2023-04-29:
**Describe the bug**
Get useless warning triggered whenever printing multiple objects. It is doubly useless because it fires when printing multiple objects 'by layer' when there is zero chance of collision. And in this particular case it's triply wrong because the second object is so short there's zero chance of collisions with the print head/rods even if I were printing in 'by object' mode. Given this warning always fires, all it means is that I am being trained to ignore it. So when it could be useful to me, I'll likely miss it.

[image]
https://www.printables.com/model/404447-flying-night-dragon

bambu123, 2023-05-04:
You may merge the parts into one object.

AdrianGarside, 2023-05-04:
> You may merge the parts into one object.

That’s missing the point. I get this error seemingly almost every time I print multiple objects when there is no issue and nothing needs to be changed. So I’m being trained by the software to ignore the error because it’s, in my experience, always wrong.

closed 2023-12-21 by tangketan