bambulab/BambuStudio #1719: Remove the limit for layer height

julie777, 2023-04-29:
**Is your feature request related to a problem? Please describe.**

P1P using .4mm nozzle and Bambu Slicer will not let me set layer height to .3 which is very common for a .4 nozzle. Instead it limits the layer height to .28.

**Describe the solution you'd like**

Allow me to set the layer height that I choose. The print profiles provides show me what BL thinks is a good setting. I get that. However, there are many reasons to choose a different height.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

MadMax389, 2023-04-30:
You can set whatever layer height you want on the quality tab regardless of the user profile selected. The print speed will still be regulated by whatever flow rate setting you have in the filament profile.
[image]

julie777, 2023-04-30:
That is not my experience, but now I that I went back and checked, it must have been added in Orca.

[image]

@MadMax389  Thanks for setting me straight.

MadMax389, 2023-05-01:
Actually, in Orca, you can raise the upper layer height limit in the printer profile.
[image]

julie777, 2023-05-01:
Well, I just got the error in BS. I have a .4 nozzle and I am using the .4mm profile.
When I tried to change the layer height to .3, this error popped up.

[image]

It did leave the .3 in the setting though.

I looked at the printer settings and it does have min and max layer height and max is set to .28. So changing it is not a problem. 

The error message needs to be reworded to indicate that the layer height is not between the min and max in the printer profile being used.

julie777, 2023-05-01:
I spoke too soon. It did let me put the .3 for layer height and continue on with other settings, but it would not allow me to slice until I went and change the max layer height in printer settings.

SaltWei, 2023-07-11:
[image]
Slice OK in v1.7 version.
Thank you very much for suggestions!
Please let us known if still has any other issue.

closed 2023-07-11 by SaltWei