bambulab/BambuStudio #1721: Text field not fitting in the Window of Device view

Laserstrahl, 2023-04-30:
**Describe the bug**
I'm using the german installation of bambu studio version 1.5.0.61
When I go to the "device" tab/page there are information on the right, unfortunately the frame on the bottom right seems to be cut off. It is not fitting the windows size. Changeing the windowsize doesn't help at all. Also a button is cut off.

Please see the Screenshot.

I was sure that this bug was already reported, but I didn't find the issue ticket, so I gonna report it.

**3mf File for This Bug**
N/A

**To Reproduce**
Steps to reproduce the behavior:
1. Start Bambu Studio
2. clock on "Device"
3. Have a look on the bottom right frame.
4. See error

**Expected behavior**
The Frame should completely fit into the windows. 
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 1.5.0.61
[image]

walterwongbbl, 2023-05-06:
Hi.
The translation for some languages is too long, and we will optimize the display for this issue.

walterwongbbl, 2023-10-13:
Please download and use the latest version of Studio

closed 2023-10-13 by walterwongbbl