bambulab/BambuStudio #1723: How Do We Add Information To The Project Screen Now? 1.6.2.4 PB 2

MadMax389, 2023-04-30:
**Describe the bug**
You used to be able to add photos, set the cover photo, add pdf’s and links. I don’t see any dialog to add information and dragging photos onto the screen doesn’t do anything.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to project screen
2. Now what?

**Expected behavior**
Ability to add information

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Win 11
 - 1.6.2.4

sisuwest, 2023-05-05:
Same issue here on macOS. Worked on previous builds to add photos and instructions.

walterwongbbl, 2023-05-06:
Hello.
the functions in the project module need to be redesigned and are temporarily unavailable for use.

MadMax389, 2023-05-06:
Thanks for the update!

closed 2023-05-18 by walterwongbbl

sisuwest, 2023-05-18:
You closed this out, but it's not fixed. Even if they need to be redesigned, there was no need to remove them in this build. You should have kept that functionality in the latest build and then released the new one with another update. I use the Project function quite often and share those files accordingly. Heck, even your winter cabin file from Bambu was done that way a few months ago.

sisuwest, 2023-07-07:
This is marked as fixed on version 1.7 and yet I still can't use the Project screen. "No Model Information"