bambulab/BambuStudio #1724: Base of support managed by brim settings, not support settings

bdebyl, 2023-04-30:
**Describe the bug**
Adjusting base of supports to object gap has no distinct setting, however is affected by the brim-to-object gap setting though it is not a brim.

**3mf File for This Bug**
Not applicable

**To Reproduce**
Steps to reproduce the behavior:
1. Import an object that requires supports near it's base
2. Slice (with supports)
3. Adjust brim-to-object gap in **Other** tab
4. Slice and observe change from brim-to-object gap

**Expected behavior**
There should be a specific setting for **support** base to object gap (there is already an xy distance to object setting but this does not change the **base** of the support.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Any
 - Version 1.5.0.60 (or latest)

tangketan, 2023-05-05:
It's not a bug, but a designed feature. The first layer of the supports are treated as brim, so affected by the brim settings. Can you give an example where this design fails?

tangketan, 2023-08-02:
No reply in a long time. Closed.

closed 2023-08-02 by tangketan