bambulab/BambuStudio #1727: Better shading of pure black filament

megamaniac, 2023-05-01:
When selecting black as filament type, a model becomes very difficult to view, especially when sliced.

It would be great if we could turn the environment lighting up, so that dark filaments are easier to see

Screenshot of black model attached

[image]

Slimmage, 2023-05-06:
I would like to add / comment this becomes especially troublesome when using the write text function - as text displays black as well, and you cannot see it against the surface. white outline on text during placement would solve. thank you

Haidiye00, 2023-05-08:
Before modification：
[image]
After code modification：
[image]

May I ask if you are satisfied with this modification? Looking forward to a reply

Haidiye00, 2023-05-08:
Please click on the image to open it

Haidiye00, 2023-08-15:
Version 1.8 will fix this issue

closed 2024-01-16 by Haidiye00