bambulab/BambuStudio #1728: Bambu Studio is not loading any of my custom profiles

julie777, 2023-05-01:
**Describe the bug**

After upgrading to the latest version, when I open Bambu Studio and select any of the profile lists (filament, print, process) none of the profiles that I created show in the list.

When I use the help menu to open the config directory my profiles are there under user/default just as they should be.

I looked in the debug log and cannot find any entries associated with loading custom profiles.

I would like to know what to try next.

(I also found a couple other issues in the log that I will create separate issues for.)

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

julie777, 2023-05-01:
I just had an idea and ran the setup wizard, even though it has been setup for months. Now all my custom profiles are visible again.

Should running the setup wizard be required after every update?

MadMax389, 2023-05-02:
Usually you just have to log out of Studio, then log back in. As long as auto-sync is selected you should get them back.

[image]

bambu123, 2023-05-04:
From this version, it does not load the default user presets and the login user presets together.

bambu123, 2023-05-22:
#1840

julie777, 2023-05-25:
It seems to have been fixed.

bambu123, 2023-05-25:
OK, thanks.

bambu123, 2023-07-11:
Close as fixed in the 1.7 beta version.

closed 2023-07-11 by bambu123