bambulab/BambuStudio #1735: Non-fatal errors on launch (Linux beta)

jcorbin, 2023-05-02:
Using 1.6.2 (beta 2) fedora app image on arch, but same has been present in earlier betas, and even when I took the time to build my own appimage from git.

The first one is a rather generic "Error running JavaScript: Unsupported result type" alert dialog; console output doesn't seem to have any additional info:
[image]

After dismissing the first one, a second instance presents:
[image]

I'll also sometimes get a "Can't find variable: wxOut0" error, but not on every launch:
[image]

At any rate, bambu studio continues to function correctly after dismissing these errors, so I'm not sure how important they are, or if I'll find something not working eventually

bambu123, 2023-05-08:
#1701

bambu123, 2023-07-11:
Close as fixed in the 1.7 beta version.

closed 2023-07-11 by bambu123