bambulab/BambuStudio #1737: Fixed IP Bambulab X1 Carbon and P1P

CHiEFROKKA, 2023-05-02:
Hello,

where can you enter a fixed IP in the Bambulab X1 Carbon? A reverse resolution is not useful in larger networks.

Currently, you can only assign the IP via the MAC via DNS server. 
This is not so great for 3D print farms. Currently, this is the only thing standing in the way of a larger quantity purchase. Maybe someone can help, how to assign a fixed IP to the printer via script or webgui.

Many thanks in advance.

sodachen22, 2023-05-05:
1. X1 only supports getting IP from DNS server, there is no way to set a fixed IP address.
2. X1 supports broadcasting its IP address through SSDP, Bambu Studio can discover the printer.
Can the two features satisfy the requirement?

Robbe-B, 2023-05-06:
@sodachen22,
 You probably mean getting an IP from a DHCP server right ? 
Not a DNS serer ?

sodachen22, 2023-05-06:
> You probably mean getting an IP from a DHCP server right ?
Yes, it is a typo. It shall be DHCP,

Patrick-UK, 2023-05-20:
The way round this is to set always use this IP in your network controller for this device.

longkeeler, 2024-01-27:
Has the status of setting a fixed IP changed since? It looks pretty ridiculous to barr the path to the simplest imagineable setting (the usual stuff, (fixed) IP, gateway (if at all)). 
What speaks against a simple upgrade? 
Here, different instances are assigned the role of a local DHCP from time to time. Having a framework of devices with fixed IPs is the safest you can get, workflowwise. Send all the cloudies back to where they came from :) 

I was short before releasing an order for a first P1S, but I'll postpone that now. 

Cheers, 
Longkeeler (longkeelers are sailing vessels famous for their course-steadiness and seaworthiness...)