bambulab/BambuStudio #1738: In Windows (10) switching to a different user while Bambulab is open will disable Dark Mode

pmcelreavy, 2023-05-02:
**Describe the bug**
In Windows, switching to a different user while Bambulab is open will disable Dark Mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Open BambuLab
2. In Preferences, enable Dark Mode
3. Switch to a user in Windows (possibly a user that doesn't use Windows in Dark Mode)
4. Switch back to original user in Windows
5. BambuLab is no longer in Dark Mode and the setting is reverted.

**Expected behavior**
Dark Mode should not be changed

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 1.5.0.61

RyanWor, 2023-05-13:
This issue also occurs when connecting to a Windows 11 system via remote desktop (RDP), however not only is dark mode disabled in preference, the interface gets stuck in this half light half dark mode. Tested this on multiple Windows 11 PCs and get the same results everywhere. 

[image]

Looks very similar to this since closed issue on macOS: https://github.com/bambulab/BambuStudio/issues/1030

rashuge, 2023-05-24:
This also occurs in Windows 11 where you simply lock the screen of the current user and then log back in.  If I leave my machine and it locks, when I login I get the half dark and half light as the screenshot shows.  The dark mode preferences is unticked.

[image]

walterwongbbl, 2023-05-29:
Hi.
We are trying to fix this issue and thank you very much for your feedback

walterwongbbl, 2023-10-13:
Please download and use the latest version of Studio

closed 2023-10-13 by walterwongbbl