bambulab/BambuStudio #1740: Settings files added to user directory are not shown

julie777, 2023-05-03:
**Describe the bug**

I added a new filament file to my user settings directory  ...user/default/filament/new.json
The ini and json file were created using Orca Slicer calibration. 

When I open Bambu Studio and try to select the filament settings then are not shown in the list.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new filament profile not using Bambu Studio
2. put the files in the user's filament directory
3. open Bambu Studio and try to select the new filament
4. It is not shown

**Expected behavior**

I expect Bambu Studio to scan the user directory for all settings files and show them in the appropriate list (printer, filament, process).

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

bambu123, 2023-05-04:
You may use the import menu to add it.

bambu123, 2023-05-08:
@julie777 Have you had any problems with user presets yet?

julie777, 2023-05-08:
@bambu123 I would have never even thought that I needed to import settings. 

I just tried adding a new pair of settings files to the user settings and then opening Bambu Studio. The settings were in the list. I have upgraded to 1.6.2.4 since creating the issue.

closed 2023-05-08 by julie777