bambulab/BambuStudio #1741: Settings for the height modifier are not saved in the project.

julie777, 2023-05-03:
**Describe the bug**

When I open a project that had a height modifier set on an object, there is no longer a height modifier.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Select and object and add a height modifier
2. save the project
3. open the project
4. height modifier is gone

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

lanewei120, 2023-05-04:
which version are you used when saving the "Height Range"
we just released a new version before ("https://github.com/bambulab/BambuStudio/releases/tag/v01.06.02.04"?)
it should fixed this issue

closed 2023-05-08 by LiZ-Li-BBL

julie777, 2023-05-08:
@lanewei120 I was probably still using 1.6.0.65