bambulab/BambuStudio #1744: AMS filament type and color selection issue

TriskelionTech, 2023-05-04:
Crossposted from [https://forum.bambulab.com/t/ams-filament-type-color-selection-issue/11957](url)

**Describe the bug**
I found a glitch in the filament selection setup on the new AMS firmware.

If I select on the printer that a filament is manufacturer “Generic” and type “PLA - Silk”, then it converts it to Bambu PLA - Silk when loaded in Bambu Studio. There isn’t even an option to select “Generic PLA Silk” on the AMS screen in Bambu Studio, even though there IS an option for that in the project filament selections.

Also seems like the new interface takes away the advanced color picker, which is a probelm for me as it does not give enough options for colors (If I have brown, beige, silk gold, silk copper, orange, and yellow all loaded, it’s basically impossible to pick “different” colors for all these on the AMS, even in Studio)

This is then compounded by the fact that even when selecting “PLA - Silk” the quick view displays on the printer HMI and Bambu Studio just say “PLA” which is frustrating to say the least. 4 extra characters (or two lines, one with “base” material type, and one with subtype would be awesome)

**3mf File for This Bug**
Not slicing related

**To Reproduce**
Load filament in any AMS slot
Set filament manufacturer to "Generic" and type to "PLA Silk"
View loaded filament information in Bambu Handy - it is CORRECT in this view
View loaded filament information in Bambu Studio - it is NOT CORRECT in this view. - Studio shows the type as "Bambu PLA Silk"
Attempt to set filament type of AMS slot to "Generic PLA Silk" in Bambu Studio - the option is NOT AVAILABLE
If many very similar filaments are loaded (i.e. Gold, Bronze, Orange, Yellow, Brown, Tan, Copper...) there are also not enough color options anymore. It used to be possible to assign additional arbitrary colors from Studio, but this seems to have been removed.

**Expected behavior**
I would expect the list of filaments available to match between Studio, Handy, and the physical machine.
Additionally, I would expect to be able to select many more color options in order to properly differentiate when multiple similar filaments are loaded. This feature was present prior to the most recent update and was removed.

**Screenshots**
AMS selection on printer - Black Generic PLA Silk
[image]

AMS selection on Bambu Handy- Black Generic PLA Silk
[image]

AMS selection on Bambu Studio - INCORRECT - Black Bambu PLA Silk
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Bambu X1C firmware 01.05.01.00
 - AMS Hub Version 00.00.00.42
 - AMS version 00.00.06.32
 - Studio version 1.6.0.65

TriskelionTech, 2023-05-07:
Heres some more detail : Crossposting from Bambu Forum:

Now that I’m back at my computer, here are the various dialogs in Studio. You can see that while it is possible to select silk in the project interface, it is NOT possible to select it in the AMS interface, which leads to the wrong profiles being used if “resync AMS filaments” is used.

Here’s the filament selection page:
[image]

Here is the filament list when manually configured:
[image]

Here are the three silk filaments, as set on the printer itself (viewed through Handy)
[image]
[image]
[image]

Here is what those look like on Bambu Studio in the AMS options area
[image]
[image]
[image]

If I use the “Resync AMS” tool, there is what is displayed. Note that although Handy and the printer itself correctly show Generic PLA Silk, Studio converts this to Bambu PLA Silk.
Before resync all button pressed:
[image]

After Resync All:
[image]

StoneLiBambulab, 2023-05-22:
same as https://github.com/bambulab/BambuStudio/issues/1688

we will fix the bug

Thanks!

Stone Li

closed 2023-10-13 by walterwongbbl