bambulab/BambuStudio #1745: Transparent / translucent material type listing

TriskelionTech, 2023-05-04:
**Is your feature request related to a problem? Please describe.**
It would be great to have "Transparent" or "Translucent" as options under the material types (such as "PLA Transparent" or "PETG Translucent") - This would provide more detail when selecting materials for printing and setting up models. Bonus points if that could also impact the auto-calculation of purge volumes (Transparent / Translucent materials often take far more than their opaque colored equivalents)

**Describe the solution you'd like**
An additional material type in the materials list for "Transparent" in each material category where these filaments are offered (or at least the major ones like PLA, PETG and PC). Best if this solution would be factored in to the automatic purge calculation to improve purge volumes on transparent materials.

**Describe alternatives you've considered**
I currently set "transparent" materials to their base equivalent, and then set the color to the lightest possible shade of their base color, to at least help the auto calculator. Usually for clear filament or very light color translucent filaments I set it to white, then add 100 to the calculated purge volumes.

zhimin-zeng-bambulab, 2023-08-18:
We have supported automatic flush calculation for transparent and translucent materials, you can have a try in the latest version

closed 2023-08-18 by zhimin-zeng-bambulab

nihilisticzebra, 2024-02-09:
> We have supported automatic flush calculation for transparent and translucent materials, you can have a try in the latest version

is there a way to choose transparent as a color on PC?

zhimin-zeng-bambulab, 2024-02-18:
> is there a way to choose transparent as a color on PC?

Currently, transparent materials can only be identified through AMS.

nihilisticzebra, 2024-02-18:
I don’t see it an a “color” option on my AMS

zhimin-zeng-bambulab, 2024-02-19:
[image]

When your filament is transparent, the filament can be automatically set to a transparent color through ams recognition.