bambulab/BambuStudio #1748: [Bug] Upgrade Installation would be blocked upon cannot overwrite "BambuSource.dll" because Microsoft Teams and Skype would load the file

ccl13, 2023-05-04:
**Describe the bug**
Recent upgrade Installation was blocked upon it cannot overwrite "BambuSource.dll". Upon check, Microsoft Teams and Skype loaded the file
```
C:\Users\<username>\AppData\Roaming\BambuStudio\plugins\BambuSource.dll
```

Closing Teams and Skype and click "Retry" made it work.

**3mf File for This Bug**
N/A

**To Reproduce**
1) Install a previous version of Bambu Studio, and install the plugin (I don't know which) that would drop `BambuSource.dll`. In my case, v01.05.00.61 was the previous version.
2) Start up Skype or Teams.
3) Download and install a newer version of Bambu Studio.
4) At almost the end of installation, an error message would show up saying the installation cannot overwrite `C:\Users\<username>\AppData\Roaming\BambuStudio\plugins\BambuSource.dll`.
5) Close and exit Skype and Teams.
6) Hit retry and it would finish installation.

**Expected behavior**
Skype and Teams and any other program should not load `BambuSource.dll`. If there's any vulnerability within the DLL, it could be used as an attack factor to gain data or control in any application that loads it.

**Screenshots**
(Sorry that I don't have one for the installation error message right now.)
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 22H2
 - Version: 22H2 19045.2913

bambu123, 2023-05-05:
It is registered as the source filter of the Direct Show framework to use the Windows Media Player for live view on Windows.
Some players using Direct Show would load this DLL.

We have a plan to drop the Windows Media Player but it will take a long time.

ccl13, 2023-05-05:
Thanks that makes sense. Meanwhile we probably should have a notification in the installer to mention this so user won't be confused.

DrivebyCoding, 2023-08-04:
Additional note since I just hit this problem: It's also opened by Nvidia Broadcast, which means it's locked from the second you log in to Windows if Broadcast is enabled. Closing down Broadcast will let you continue the install.

IDontEatSoap, 2023-08-07:
Same issue with NVidia Broadcast using the DLL. Download ProcMon and search for Bambu, it should show you what's using it.

bambu123, 2023-08-07:
Install BambuStudio 1.7.2 and later versions will hide BambuSource.dll from these Apps.