bambulab/BambuStudio #1750: Global flag to discard unload filament when print complete

projectcobalt, 2023-05-05:
**Is your feature request related to a problem? Please describe.**
Continuous improvement of UI/UX, including sequential printing of files and general pre-start time improvement.

**Describe the solution you'd like**
provide a global flag to allow printer to not unload filament when print is complete. this can either be done in the "Send" box where filament/pre-start functions are selected or inside global AMS settings.

**Describe alternatives you've considered**
Other than exploring g-code no other alternative... this will improve Out of Box experience

**Additional context**
Add any other context or screenshots about the feature request here.

bambu123, 2023-05-08:
It this relate to AMS?

projectcobalt, 2023-05-20:
It could be related to AMS directly if option was global to AMS settings, or it could per print selectable during the send to printer dialogue box

projectcobalt, 2024-01-25:
Will this be added into the feature enhancement roadmap?