bambulab/BambuStudio #1754: Side Spool

Benderrodrigu3z, 2023-05-05:
Every time I go to print TPU or something I have to mess up one of the slots in BambuStudio and tell it change that line to TPU so that I can create a correct slice file then print then switch it back constantly, can we get a side Spool Slot? the Sync could even do a pull info on that line from the machine

bambu123, 2023-05-08:
Do you meen this that already exists.
[image]

Benderrodrigu3z, 2023-05-08:
> Do you meen this that already exists.
> [image]

No I mean what I said, in the BambuStudio slicer, your forcing me to mess up a filament slot to TPU only to change it back after printing

[image]

bambu123, 2023-05-08:
Are you want the seventeenth slot after syncing the AMS filament list?

bambu123, 2023-05-08:
#1698

Benderrodrigu3z, 2023-05-08:
[image]
more like this, plus it would work with those without an AMS