bambulab/BambuStudio #1755: Bambu Studio Crashes Slicing Large Prints Ubuntu Appimage 1.6.2

mltriebe, 2023-05-05:
**Describe the bug**
When I try to slice large files in Mint 21 the program stops slicing at some point in the process.  When it does I can't move my mouse or anything and I have to restart the computer.  Therefore I do not have a .3mf file to share

**To Reproduce**
Load a large file and slice it.

**Expected behavior**
I expect the file to finish slicing.

**Screenshots**
I can't take a screenshot, the computer becomes unusable.

**Desktop (please complete the following information):**
 - OS: Linux Mint
 - Version 21.1

lanewei120, 2023-08-22:
hello @mltriebe 

I am sorry we missed this issue before.

do you still have this issue?
if yes, is it a 100% produciable issue or not?
if you slice a small stl, did this issue happen?
and could you share your stl, we can have a debug then

SaltWei, 2024-01-18:
Hello @mltriebe  Does this issue still exist under your PC environment?