bambulab/BambuStudio #1756: Ubuntu Appimage 1.6.2 Adaptive Layer Height Does Not work

mltriebe, 2023-05-05:
**Describe the bug**
When I load any file and try to use adaptive layer height the program freezes.

**3mf File for This Bug**
I don't have a .3mf file, the program freezes.

**To Reproduce**
Steps to reproduce the behavior:
1. Load any file.
2. Click on adaptive layer height.

**Expected behavior**
Adaptive layer heights to work.

**Desktop (please complete the following information):**
 - OS: Linux Mint
 - Version 21.1 Cinnamon

cmguo, 2023-05-19:
[image]

Do you mean clicking here?

I can't reproduce it here. Can you help provider the freeze stack using gdb?

mltriebe, 2023-05-19:
I tried a live disc on another computer and it worked fine. The one that had this issue that pop up would not come up to select adaptive or smooth. Feel free to close this if you want to.

bambu123, 2023-05-22:
@mltriebe Is your other computer also the Linux system?