bambulab/BambuStudio #1760: GCode Path Error because of Collision

Nosnbora, 2023-05-05:
Because Bambu Studio has no Option for Support Brim I used a flat Cylinder on which the Tree Support is printed. 
This is technically a collision because the Support is printed inside the cylinder, but in real life, it works fine. 

Now with the new version of Bambu, you took that option from me because I get the Error "Conflicts of gcode paths have been found at layer 0" and I can't print it. And also you still didn't add the support brim feature. Like WTF Bambu Lab? Are you working against the Customer?

Please fix this mess. Don't try to make everything too safe... it makes things worse sometimes. 
[image]

nahashrishon, 2023-05-09:
I'm also having this issue with my files. Keep getting the conflict at layer 0 z height 0.00mm.

reduced it down to the first layer and everything seems fine 
[image]

if I delete all the objects but one and then slice it, still gives me the above error but allows me to print

[image]

faulty file is attached 
[unchecked.zip](https://github.com/bambulab/BambuStudio/files/11426773/unchecked.zip)

pstagman, 2023-05-09:
I'm getting this same ERROR from a file that's been working perfectly for months. Ince upgrading to Studio V 1.6.2.4

[image]

tangketan, 2023-05-12:
@nahashrishon If you look closely you'll see there are indeed path conflicts for multiple copies. If you remove all except one, the conflicts are gone, and you may notice the "Print Plate" button is back. It's just the message didn't disappear, which is a bug and we'll fix it in the next version.
[image]

Besides, I agree that the conflicts of support paths aren't as severe as that of object paths. In the next version, we have increased the threshold to allow a larger amount of support path conflicts.

tangketan, 2023-05-12:
@Nosnbora @pstagman Please share your 3mf files so I can tell if it's a bug or not. Most of the time there are indeed path conflicts, maybe too small to notice.

magic-happenz, 2023-06-03:
@tangketan its actually quite simple to reproduce. Simply make a cube with cylinders at the corners and you will see. I would also love to share a 3mf file with the same which ultimately fails to slice in a way you can print it but you can not share files here. If you can not reproduce please propose a way to share the file.
[image]

I remember in an earlier version it sliced it like in [this](https://github.com/bambulab/BambuStudio/issues/1393) feature request.
If the request is not going to worked on soon it would be great to at least allow again to print these overlaps.

clowrey, 2023-07-22:
There NEEDS to be an override for this "error" or just convert it back to a warning!!! I cannot print a model that has overlaps on purpose with another model - it prints fine and is intentional!! - I have to revert to the previous version to print this file! please either provide an over-ride check box somewhere in the settings or just make this a warning instead of hard error that prevents printing! 
Thanks :)

tangketan, 2023-08-02:
We understand your needs and have changed the error to a strong warning in the latest Studio 1.7. 
[image]

closed 2023-08-02 by tangketan

terryb58, 2023-10-19:
This is not true for 1.7.7.89. As a matter of fact, it is much worse now. Printing flat objects that the hot end would not collide with still need a wide margin. See images. 
[image]
[image]

It should be possible to print them this close and still be in no danger of the hot end colliding with the models.
[image]