bambulab/BambuStudio #1761: Bambu Studio doesn't actually reach edges of the screen

zswickliffe, 2023-05-05:
**Describe the bug**
When in full screen, the program leaves gaps around the edges where the desktop is clearly visible.

**3mf File for This Bug**
N/A

**To Reproduce**
Steps to reproduce the behavior:
1. Open Bambu Studio
2. Click "maximize window"
3. See the gaps surrounding the program window

**Expected behavior**
I would expect the program to reach the extents of windows when maximized.

**Screenshots**
On my main monitor, there are gaps on all sides: 
[image]

On my secondary monitor, it only gaps on the right side (the shared side between the monitors):
[image]

My monitor configuration:
Main monitor: 
[image]
Secondary monitor:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 Pro
 - Dual monitor setup, the main monitor is 3840 x 2160, the secondary monitor is 2560 x 1440. Screenshots of their configuration above.
 - Version	10.0.22621 Build 22621

bambu123, 2023-05-08:
#1668

cmguo, 2023-05-15:
@zswickliffe
We can't reproduce it here.
Could you please show me the window size value in the config file?
The config file can be located from the menu "Help > Show Configuration folder".
Make sure to minimax the window and close Bambu Studio before reading values from the config file.

[image]

bambu123, 2023-05-15:
@zswickliffe
We can't reproduce it here.
Could you please show me the window size value in the config file?
The config file can be located from the menu "Help > Show Configuration folder".
Make sure to minimax the window and close Bambu Studio before reading values from the config file.

[image](https://user-images.githubusercontent.com/46235824/238252731-85099a31-183e-4b04-bf8b-1604ec9fc59f.png)

zswickliffe, 2023-05-18:
@bambu123 sure thing:

On my main monitor (4k):
[image]

On my secondary monitor (1440p):
[image]

zswickliffe, 2023-06-05:
@bambu123 need anything else from me?

bambu123, 2023-06-05:
No, thank you. We will fix it in the next version.

closed 2023-07-11 by bambu123

only1chip, 2024-06-12:
I have a related problem.  When I launch Bambu Studio on my triple-monitor setup (all screens are portrait (@ 1080 wide) - the max width to which the BStudio window will stretch is 1080px.  Changes to the "window_mainframe" values in conf file don't seem to have any effect.  The only solution I've found is temporarily setting one of my screens to landscape while I launch BS.  Then I can revert my monitor settings for portrait, and the BS window will scale to a more convenient size (ie a size where all the window components are visible without horizontal scrolling).  Running BS ver 1.8.4.51 on Win 10 Pro 10.0.1.19045