bambulab/BambuStudio #1763: BAMBU' STUDIO SI BLOCCA

Raffone70, 2023-05-06:
**Describe the bug**
Bambu studio dopo qualche utilizzo si blocca la versione è la 1.6.2 e non c'è modo di ripristinarlo..

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

bambu123, 2023-05-08:
Please send us the logs from the menu 'Help > Configuration Folder'.
If it is reproducible, a detailed video showing the steps would be best.

Raffone70, 2023-05-08:
Grazie, per la celerità,

praticamente non ho fatto nulla di particolare, intanto spesso nel fare lo "slice del piatto" si blocca nel senso che la schermata rimane bianca e va in loop. si deve uscire dal programma riavviarlo e quando si fà lo slice funziona.

l'altro giorno invece inviando alla stampante, la stampa si è bloccato bambù studio e ho dovuto disinstallarlo e riinstallarlo!

a presto

Raffaele Greco

bambu123, 2023-05-08:
Have you started the live view which is known to hang on the system which has dual graphics cards?
Whenever it is locking, please use the Windows Task Manager to create the minidump of BambuStudio, the dump is useful to analyze the hung problem and please send it to us.

Raffone70, 2023-05-08:
La visualizzazione del video della camera dal bambù studio si l'ho avviata e funziona quasi perfettamente, io sul pc ho due schede video ma penso ne usi una sola, credo quella configurata che poi è quella non della scheda madre. (è una quadro da 2 gb)

come si fa la procedura che mi avete suggerito? per il minidump?

cortesemente

Raffaele Greco

bambu123, 2023-05-08:
You can create minidump with the following steps:
- Open the Windows Task Manager
- Find and select the Bambu Studio process in it
- Right click the process and choose the 'Create Minidump File' menu item
- Select a folder to save the file

Raffone70, 2023-05-08:
ho eseguito la procedura, ma mi dice di aver creato un filw da 1,6 gb,

come faccio a inviarvelo?

Raffaele G.

bambu123, 2023-05-08:
Can you help zip it and send it via your Dropbox?

Raffone70, 2023-05-08:
Posso inviarlo tramite WETRANSFER?

bambu123, 2023-05-09:
Yes, please!

bambu123, 2023-05-15:
Sorry for being late. But we can't confirm the block reason from the minidump you provided last week although it does show something. 
Could you please send us more dumps in the blocking situation?
See if you can search for more dump files from the Windows system folder "%APPDATA\..".

bambu123, 2023-06-15:
@Raffone70 Are you still facing this problem?