bambulab/BambuStudio #1765: Unable to recover failed print from a given layer height

Kaloszer, 2023-05-06:
**Describe the bug**
Edited gcode, which has everything edited out to start from a certain layer, still tries to use AMS if plugged in, and attempts to do bed level when that is not in the g-code.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a sliced file, start printing. Abort print as soon as it starts
2. Grab the plate_1.gcode
3. Edit out an 'n' number of layers, leaving just the initial temp settings
[plate_1.txt](https://github.com/bambulab/BambuStudio/files/11412480/plate_1.txt)
NB! had to snip the middle due to size limit in github, so this file is not '''printable''', but you should get the gist.
5. Printer still tries to do bed level where it is not stated anywhere in the gcode

**Expected behavior**
Printer should just set the temp and go from given layer instead of attempting to use AMS, or bed level.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version N/A - printer - p1p latest firmware as of writing 5/6/23

Kaloszer, 2023-05-06:
Other possibility is embeeding such a feature in the slicer, eg. right click on slicing timeline -> start printing from this layer height.
That would be the best UX :)

bambu123, 2023-05-08:
Resume from failure is difficult, but what kind of failure have you meat?

Kaloszer, 2023-05-10:
> Resume from failure is difficult, but what kind of failure have you meat?

Filament runout, AMS couldn't pull out a glued spool end. Not relevant to the issue at hand.

blackr2d, 2024-07-24:
I second that. I just had a clogged nozzle after 18 hours of printing in... PA12-CF...

It would be really fantastic to be able to find the layer to restart and "start from layer x" in the slicer.

I will end up printing the remainder separately and gluing, but that may compromise the mechanical properties of a critical component...

gaolegao-lx, 2024-07-25:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

blackr2d, 2024-07-25:
> **Please upload log through Handy APP, and give us your ticket number**
>
> _**Upload log files through Bambu Handy**_ Log files from the Bambu printer can be uploaded via Bambu Handy over the network, [...]

Thank you  but I've printed several parts since then and I'm not sure which one it was.
I have printed the missing layers and glued the object together, it was aligned so all major stress lines were avoiding layer lines, so glue adhesion seems to be sufficient in this case.

I'm not sure what support ticket can change here.

In the future it would be really good to find the last printed layer and "start from layer X" in slicer, in my case it would have just continued the print after I've cleared up the clog.

Thank you for your consideration, looking forward to "start from layer X" option when available.

blackr2d, 2024-07-25:
As luck would have it - I started printing an AMS riser in Marble PLA this morning and got the same issue, I've lodged a ticket with logs under US240725474001.

Thank you in advance @gaolegao-lx !