bambulab/BambuStudio #1766: [Request] Fill bed with instances

BambuLabDiscord, 2023-05-06:
Prusa Slicer has an option to Fill the bed with instances in the context menu when you right click on a model. I would love it in Bambu Studio

MadMax389, 2023-05-07:
+1 for this. Also add true instancing so that a change to one model on the plate changes them all.

tangketan, 2023-08-02:
This feature is included in Studio 1.7 now.
True instancing is intentionally removed because it's too complicated and seems useless, but we can discuss.

MadMax389, 2023-08-04:
@tangketan While it may be complicated, I respectfully disagree that true instancing is useless. Making a small change (like touch up painting, adding a negative volume, changing z-height) on a plate full of the same model requires deleting all but one model, cloning again and rearranging on the plate again. There are open requests for this since last year like #1817 and #496. As detailed in #496, the option exists in Prusa Slicer and many folks would like to see it implemented in Studio.

tangketan, 2023-08-10:
I understand your need. Our team is discussing adding this feature in the next version. 
I just changed the title a bit to follow this issue easier.

man55, 2023-08-14:
waiting for this feature to be added

morganvenable, 2024-07-08:
Where is this feature?  It's horrible to try to maintain any production workflow in BambuSlicer, I'm really running out of working time to manage "clones" that don't correctly maintain identical support painting, etc.

morganvenable, 2024-07-08:
@tangketan ^