bambulab/BambuStudio #1768: Cubic infill have layer gaps

DzzD, 2023-05-07:
It seems that now, maybe due to latest release of the studio, there is empty line, at least on cubic infill.

Here are screenshot of cubic inifill : 

[image]

**To reproduce :** 
On Bambu Studio, add primitive cube, set its infill to cubic 20%

QingZhangBambu, 2023-06-09:
It concerned how  GCodeviewer draw the shape of the line. The printing artifact is fine.
[image]

DzzD, 2023-06-10:
Ok I ll check that, but why it happen only for cubic infill and why now ?

DzzD, 2023-06-10:
I can confirm that the print is ok, but still disappointing that the preview is not as good as for other infill

closed 2024-01-16 by SaltWei