bambulab/BambuStudio #1770: 1.6.2.4 does not start

ffafe5, 2023-05-07:
I accepted the prompt from my previous version to update to 1.6.2.4. Now the software fails to start up. I get the basic frame with the file bar and then the program hangs. I tried reinstalling but the same occurs.

Windows 10, 64bit

ffafe5, 2023-05-07:
I tried going back to 1.6.0.65, which I figure must have been my previous version or close to it since I installed the software 2 weeks ago, and it still won't open.

ffafe5, 2023-05-07:
[image]

ffafe5, 2023-05-07:
Restarting my PC doesn't help either -- right now my $1500 printer is a paperweight.

bambu123, 2023-05-08:
Maybe there are file corruptions in the configuration folder.
Please backup the configuration folder '%APPDATA%\BambuStudio\', rename it let Studio have a refreshing start.
Please send us the old configuration file for analyze.

ffafe5, 2023-05-08:
Sorry, I already tried that before reading your message and it did work. More specifically, prior times that I uninstalled the software I did not select the checkbox for deleting user data, and finally I did select that, so I assume the appdata you're looking for is now gone. Then I reinstalled the latest software version from the Bambu website directly, and it was OK again.

Ideally the software would not fail like that in the first place, but in the end it wasn't so bad -- my biggest concern with deleting my user data was losing my custom filament profiles, which I did lose. But once I logged into my account in Bambu Studio, it offered to sync me my cloud data, which restored the app data that I cared about.

So all is well on my end. Thank you for taking interest in the logs, and sorry that they are not available.

closed 2023-05-08 by ffafe5

lifenouveau, 2023-07-12:
Same issue with 1.7.0.x beta.

bambu123, 2023-07-12:
> Same issue with 1.7.0.x beta.

What about previous versions?
Is it not responsible for mouse actions? Is there a Not-Responese dialog showing later?