bambulab/BambuStudio #1771: Why are the filament profiles for the P1P in a subdirectory now?

julie777, 2023-05-07:
Bambu Studio installs filament profiles for the P1P in a sub-directory of filaments.
```
C:\Program Files\Bambu Studio\resources\profiles\BBL\filament\P1P
```
Why are they not in the filament directory with the rest of the system filament profiles for BBL? 
- They have distinct names, so there is no name conflict.
- The other kinds of profiles (process, machine) do not put P1P profiles in a sub-directory.

I do not understand the inconsistency.
It causes problems for automated processing of profiles.

I have created an app to provide a list of changed profiles for each release. This will be used to help users who have custom profiles and/or custom gcode make sure they do not have issues when the system profiles change.

QingZhangBambu, 2023-05-26:
putting all files in one hierarchical directory will be difficult to find and maintain.

julie777, 2023-05-29:
@QingZhangBambu 
I do not agree with that, but if it were true then I would expect both the P1P and the X1 to have separate directories that are siblings.

QingZhangBambu, 2023-05-29:
@julie777 we will recognize it.