bambulab/BambuStudio #1772: Please add configurable notifications

julie777, 2023-05-07:
**Is your feature request related to a problem? Please describe.**

I have Bambu Studio running and connected to my printer. The printer notifies Studio when something happens
- filament run out
- print complete
- front cover fell off print head (I just received this even though the cover was fine)
- printer failed to parse 3mf file
- etc.

I usually start a print and the put Studio in the background while I work on something else. The printer is in a different room. Currently, if something happens I don't know about it until I check Bambu Studio to see how the print is coming along.

**Describe the solution you'd like**

I would like the ability to configure notifications so that Studio could notify me of printer events. 
I would like to be able to have them sent to the Windows notification manager.
I would like the option to have them signal with a sound.

**Describe alternatives you've considered**

Currently, I just have to periodically check by looking at Bambu Studio. I frequently find that for some reason the print has not started, or it is paused waiting for something, but with no notification I have wasted a good amount of time because I was not checking constantly.

**Additional context**
Add any other context or screenshots about the feature request here.

ExecutiveAV, 2024-01-09:
Commenting to keep track of this

malciin, 2024-11-28:
+1, new bambu owner here and I miss this feature very much from the first day