bambulab/BambuStudio #1774: Overhangs Not Detected when Support Z Distance is 0

djngnear, 2023-05-08:
**Describe the bug**

Setting the Top Z Distance to 0 in order to use support interface filaments makes the slicer ignore the overhangs therefore the printer won't slow down for them. Setting the Top Z Distance to anything above 0 makes the overhangs detectable again.

**To Reproduce**

1. Go to Support Tab
2. Click on Top Z Distance and set to 0
3. See error in slice preview of model with overhangs.

**Expected behavior**

The overhangs should be detected in the slice preview.

**Screenshots**

[image]
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows] Windows 10
 - Version [e.g. 22]

tangketan, 2023-05-15:
Please share the 3mf so I can check the problem. Thanks~

djngnear, 2023-05-15:
> Please share the 3mf so I can check the problem. Thanks~

https://drive.google.com/file/d/1NMJYv5DDirrNsJYdhf1VhGcfJvOwEec0/view?usp=sharing

tangketan, 2023-05-16:
I see. If speed is what you care, actually it's fine. You can check the speed profile, the speed of the overhang perimenters are actually lowered down.
[image]

I guess what confuses you is the displayed perimeter type. The overhang perimenters are not classified as "overhang wall" when top z distance=0. This issue is related to #1464 . The perimeter type depends on the support type. But it's OK in your case. The speed calculation is not affected by the detected perimeter type.

closed 2023-08-02 by tangketan