bambulab/BambuStudio #1775: Re-implement the search function that Prusa Slicer has.

julie777, 2023-05-08:
**Is your feature request related to a problem? Please describe.**

In Prusa Slicer you can use `<ctrl>-F` to search for a setting. Apparently Bambu Studio has removed this.

**Describe the solution you'd like**

Restore the search functionality that was present in Prusa Slicer.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

bambu123, 2023-05-08:
You can search parameters from here:

[image]

julie777, 2023-05-08:
That doesn't really work, or at least not like the search in Prusa Slicer. In Prusa Slicer `<ctr>f` searched all settings. You can type part of a setting name, such as "max", and see a list of all settings that match. 

What you indicate does some kind of very limited search. 

If I search for "max" in PS I get about 30 results. Search in Studio returns 3.
If I search for "seque" being the first part of sequential I get five results, an they even include preferences.