bambulab/BambuStudio #1776: Recent Projects list not properly updated (due to multiwindow launch ability)

bolsoncerrado, 2023-05-08:
**Describe the bug**
If you open more than one instance of BambuLab, projects opened in the later don't get to the "Most Recent opened" list.

**3mf File for This Bug**
Any file

**To Reproduce**
Steps to reproduce the behavior:
1. Open Bambulab and open any 3mf
2. Doubleclick on any other 3mf. A second instance of BL will open.
3. Close all BL instances
4. Reopen BL. Most recent list does NOT show second instance 3mf.

**Expected behavior**
Either DISABLE "multiple instances allowed"  (or make it user selectable) or properly update the Most Recent opened list.

**Desktop (please complete the following information):**
 - OS: MacOS Ventura

bambu123, 2023-05-10:
Many app configs have this problem such as filament list, and preset selection. They can not be synced between instances.

bolsoncerrado, 2023-05-10:
It's already an issue you allow multiple instances of the slicer opened. It only calls for issues (I get multiple app crashes now that I didnt have in the past (MacOS ventura) and it's since the multiinstances madness began....

Anyhow I can't understand why in the selected instance you're working on, if you enable tree supports, default can't be de-selected or a tree style pre-selected upon click. :?

catyan0930712-maker, 2026-10-11:
### Additional report: Windows 02.08.04.57 Public Beta — model intermittently missing from Recent Projects after saving

I am adding a current Windows report here because the missing recent-project history may be related to the multi-instance behavior discussed above. **The shared root cause is not confirmed.**

**Environment**
- Bambu Studio: **02.08.04.57**, verified from the installed executable, configuration, and Studio log headers. This matches the latest Windows Public Beta download checked on 2026-10-12; the .61 revision on the same release page is for macOS.
- OS: Windows 11 Pro Insider Preview, **10.0.26220**.
- Application download origin and printer/profile selected at the time of the incident: not established.

**User-observed workflow**
1. Edit a self-created model in Bambu Studio.
2. Click the save icon in the upper area above the slicing controls. The user described this visually as an “SD-card save button”; its exact tooltip/action has not been independently verified.
3. Close Bambu Studio and open it again.
4. Sometimes the model cannot be found in the home screen's recently opened model/project list.

This **does not happen every time** and cannot currently be reproduced reliably. These are the reported circumstances, not a deterministic reproduction recipe. The exact affected filename, occurrence time, saved-file type, and application close order are currently unknown.

**Actual / expected**
- Actual: the model is intermittently absent from the recently opened list after reopening Studio.
- Expected: a successfully saved editable project should remain available in Recent Projects, subject to the configured history limit. If the clicked action only exports print data rather than saving an editable project, please clarify that distinction in the UI.
- This report concerns the history entry; it does not establish deletion or corruption of the model file.

**Read-only diagnostic observations**
- Three separate Bambu Studio processes were running during inspection.
- Configuration: `single_instance = false`, `max_recent_count = 18`.
- The persisted `recent_projects` section contained 18 entries, and all 18 paths existed when checked.
- These observations make this existing multi-instance issue a useful lead, but do not prove which conditions caused the reported disappearance. The history limit and whether the button saved a project or exported print data should also be considered.

**Log collection and validity**
12 Studio logs starting on 2026-10-12 (Asia/Tokyo) were collected from `%APPDATA%\BambuStudioBeta\log` and preserved locally. All 12 had a recognizable `BEGIN_HEADER` / `END_HEADER`, parseable JSON headers, and nonempty encrypted payloads. The ZIP was read back and all 12 file hashes matched the captured copies.

Sanitized header fields shared across these logs:
```text
app_name: BambuStudio
app_version: 02.08.04.57
app_build_time: 20260922-164759
enc_block_size: 16
```

These are genuine, structurally recognizable Studio logs, but **their encrypted contents have not been decrypted or validated**, and no known incident timestamp is available to tie them to the missing history entry. They therefore do not yet confirm the cause. Raw encrypted logs, account configuration and private model files are not attached to this public comment. No project from a confirmed failing occurrence has been identified.

Please advise on an appropriate private channel for the encrypted logs and any focused diagnostics needed to distinguish a history persistence problem from save/export behavior. If this workflow has a separate cause, please let me know whether a separate issue would be more appropriate.