bambulab/BambuStudio #1777: Save Project does not work correctly for a new project

julie777, 2023-05-08:
**Describe the bug**

When I create a new project, import an STL file and then use `<ctrl>s` to save nothing happens.
There is no warning or error.
I then open the file menu and Save Project is disabled.

[image]

This is not standard behavior. The standard behavior that applications have been using for more than two decades is to have Save invoke Save As when the current document is new and has not been saved. In addition, the Save As dialog should have a default name filled in, which is commonly something like "Untitled 1".

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Create new project
2. try to save using `<ctrl>s` or File/Save

**Expected behavior**

The "Save As" dialog should appear, with a default name filled in, allowing the user to optionally give the project a name of their choosing and then save it.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

julie777, 2023-05-08:
It turns out that save does work for a new project, just not always correctly.
I was writing this bug because I discovered it while writing a different bug report. The symptoms that I saw and reported above were confusing because `<ctrl>s` had saved the file, but without displaying a dialog for me to enter a file name or choose a location.

**Works**

If I create a new project and immediately save it, the Save as dialog opens with the file named Untitled.3mf. This is working as I would expect.

**Bug**

However, if I create a new project, import an STL file and then use `<ctrl>s` it does not open the save as dialog. Instead it silently saves the new project using the name of the STL file, which is not standard, not what one would expect, and could possibly cause accidental overwrite of an existing project file.

bambu123, 2023-05-10:
I can't reproduce it. 
Maybe you are using an opened project instead of creating a new one before importing an STL file.

bambu123, 2023-05-22:
@julie777 
Is it responsible recently?

bambu123, 2023-07-11:
Close as fixed in the 1.7 beta version.

closed 2023-07-11 by bambu123