bambulab/BambuStudio #1778: Make it easier for users to submit issues.

julie777, 2023-05-08:
**Is your feature request related to a problem? Please describe.**

Users are frequently reporting issues on the community forum and Reddit.
It would be more useful if they reported them properly here.
In addition, when users do report issues on GitHub they often miss filling in the environment fields, such as OS and version. (I often forget to scroll down. Then after submitting the issue I remember and go back and fill them in.)

**Describe the solution you'd like**

1. Update the issues templates to move the environment based fields to the top so that the user will see them first and fill them in. (This helps the maintainers.)
2. Add a "Submit Feedback" button in a conspicuous place in Bambu Studio that the users will notice and use instead of going to the forums. In addition, that button could not only open the github page for a new issue, but it could also provide the environment information directly, eliminating the possibility that the user will forget.
3. Add notices on the community forum and Reddit that the community will see, telling them to report issues on GitHub.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

lanewei120, 2023-05-24:
we have refined 1
and will add 2 in following versions

and also we will ask our customer support to improve 3

lanewei120, 2023-08-05:
also add an entry in Help
[image]

closed 2023-08-05 by lanewei120