bambulab/BambuStudio #1780: Single walled prints with multiple colors fails to slice properly

ChadDevOps, 2023-05-08:
**Describe the bug**
An stl that has a one line thickness will not slice if each side of the wall is a different color in Bambu Studio.

**3mf File for This Bug**
Attached is a 3mf example demonstrating this behavior. This is not intended for print. Slice the plate to see the wall vanish.
[block line.zip](https://github.com/bambulab/BambuStudio/files/11425220/block.line.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Color an STL with thin walls, a different color on each side, and slice.
 
**Expected behavior**
Render and print the single walled line. Perhaps choose based on filament order, or external vs internal facing wall?

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 1.6.2.4
 
**Screenshots**
Screenshots attached for 3mf.

Original issue discovered on a Hex3D Grogu file when attempt to color the sleeve, black inside, brown/tan outside.

[image]
[image]
![grogu sleeve](https://user-images.githubusercontent.com/26794364/236936
[image]
352-7686a969-2932-4a3c-b77e-97dd196b1e63.jpg)

zhimin-zeng-bambulab, 2023-05-16:
@ChadDevOps 
In this case, you can avoid the problem by modifying the parameters as shown in the image below：
[image]

zhimin-zeng-bambulab, 2023-05-16:
But if the wall is too thin and only the line width, it will not be able to generate the correct wall anyway,  it is beyond the precision limit of the printer.

ChadDevOps, 2023-05-18:
Using Arachne does not appear to resolve the issue completely if the wall is colored differently on either side (no matter the width). However, with Grogu, the sleeve's inner black wall (changed to pink in screenshots below) does generate, however, the outer wall does not (I also used a more aggressive 25% minimum wall size). Previously the inner wall did not generate and was a physical gap in the print. If I paint/color more of the sleeve (e.g. increase the color depth/distance on the outside and inside of the sleeve using the same white/tannish color) the issue doesn't surface. 

Note: If the object is not painted/colored the issue doesn't surface under classic, walls are generated correctly.

Original painted sleeve with Arachne walls.

[image]

Newly painted sleeve with more depth of the light tan/white color both outside and inside.

[image]

zhimin-zeng-bambulab, 2023-05-26:
Currently, besides changing the Wall generator to Arachne, I have not found any better way to avoid this problem. Or paint the thin-walled inner and outer surfaces the same color.