bambulab/BambuStudio #1781: Plate names do not show in plate lists

julie777, 2023-05-08:
**Describe the bug**

In the objects list plates are shown as "Plate n" where n is the plate number. 
In the plates sidebar when in preview mode, only the plate number is shown.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
When a plate has a name, and it always has at least the default name of "Untitled n", the name should be shown instead of the number. It is much easier to find the correct plate to make a change or print it using the name.

Even better, if a plate has not been given a name other than the default and a single object is placed on the plate, then use the object name as the default name for the plate.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:  Windows]
 - Version 1.6.2.4

Haidiye00, 2023-05-09:
Can you provide pictures

julie777, 2023-05-23:
@Haidiye00 
As you can see, neither the plates in the object list or the plates in the sidebar are showing the names that I have given them.

[image]

Haidiye00, 2023-05-24:
[image]

This feature is currently almost developed, and the effect image is as shown above. Due to the small space, adding a name to the disk thumbnail is not considered. Still undergoing internal testing and not yet contracted. Please wait for the new version of the software. May I ask if there are any additions to the rendering? Thank you

closed 2023-05-29 by Haidiye00