bambulab/BambuStudio #1782: Error message on Ubuntu 23.04: Error running JavaScript: Unsupported result type

mzanetti, 2023-05-09:
Ever since I've upgraded to Ubuntu 23.04, launching Bambu Studio produces a popup with this error message twice:

Error running JavaScript: Unsupported result type

It doesn't seem to actually break anything tho, at least I haven't come across something related yet.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Ubuntu 23.04
2. Download latest Bambu Studio Ubuntu AppImage
3. Make executable and launch
4. See error

**Screenshots**
[image]

**Desktop:**
 - OS: Ubuntu
 - 23.04 lunar

bambu123, 2023-05-10:
Same As #1701

mzanetti, 2023-07-08:
Fwiw, this seems to have gone with the latest public beta version. There's only a fedora AppImage, but it runs fine on Ubuntu 23.04 and this error message is gone.

bambu123, 2023-07-11:
Close as fixed in the 1.7 beta version.

closed 2023-07-11 by bambu123