bambulab/BambuStudio #1783: Send to SD via Cloud

Lesani, 2023-05-09:
**Is your feature request related to a problem? Please describe.**
My printer lives on a different LAN than the PC I work from. I can print directly via the cloud service, but when I prepare a file in advance, and want to send it to the printer while the printer is currently printing, I cannot do that because the "Send" function in contrast to the "Print" function, only wants to communicate on the same network and does not use the cloud service.

[image]

**Describe the solution you'd like**
Enable sending a gcode file via the cloud service, same as direct printing

**Describe alternatives you've considered**
I have no alternative, I cannot get on the same LAN (large industrial company, not home user)

**Additional context**
Add any other context or screenshots about the feature request here.

walterwongbbl, 2023-05-10:
Hi.
Printing files can also be copied to the printer using an SD card.
Does this meet your needs?

Lesani, 2023-05-10:
No, because it is additional ways between the printer, maintaining the SD card and fetching it separately. You already have a system in place to send the GCode to the printer via the cloud. Please extend it to be also able to send a file, without directly starting a print

walterwongbbl, 2023-05-11:
Hi.
I have collected your suggestions. We will discuss this feature in the next requirements discussion

michi1g, 2023-05-16:
+1 for this feature

morganvenable, 2024-02-09:
+1.  Ridiculous that a file sent to the printer can't be stored locally for later reference.  Makes storing/selecting prints a very unnecessary hassle.

antonmorch, 2024-05-17:
+1. I faced the problem for the first time today, and wondered why it wasn't possible, since I didn't enable LAN-only mode. 
Turns out it isn't possible at all to send via cloud.

pleite, 2025-01-12:
+1 It's a mess to have to bring a PC to the printer lan, as it's another building, because we can't just load a sliced model. For example to place adaptive layers or combine parts without making a published model for handy, or just put some gcode without geometry.