bambulab/BambuStudio #1784: Sync AMS Settings via cloud

Lesani, 2023-05-09:
**Is your feature request related to a problem? Please describe.**
My printer lives on a different LAN than the PC I work from. I can print directly via the cloud service, but when I prepare a file I cannot sync my AMS settings, as this only works when the printer has a direct IP connection on the same network. The Live-Camera DOES work via the cloud, so I think syncing AMS should not be that difficult?

[image]
[image]

**Describe the solution you'd like**
Enable syncing AMS settings when not on same LAN

**Describe alternatives you've considered**
I have no alternative, I cannot get on the same LAN (large industrial company, not home user)

**Additional context**
Add any other context or screenshots about the feature request here.

Lesani, 2023-05-10:
since it contains machine and personal identification data, I have sent the file to the e-mail address in your profile @bambu123

bambu123, 2023-05-11:
Got it, thank you!

walterwongbbl, 2023-05-31:
Hi.
Will the studio continue to be unable to connect to the server.
In non-LAN mode, printer data is transmitted through the cloud. If prompted as failed to connect to the server. Unable to retrieve printer data.

walterwongbbl, 2023-05-31:
Hi.
How has the recent usage been? Is everything running smoothly?

walterwongbbl, 2023-08-02:
Please use the latest version of Bambu Studio.

closed 2023-08-02 by walterwongbbl