bambulab/BambuStudio #1786: Bambu Studio crash when using raft with small rounded surfaces

pedro-pedrosa, 2023-05-09:
**Describe the bug**
When using a raft with 0.08mm default preset and a small model with rounded bottom surface, bambu studio crashes on the slicing operation.

**To Reproduce**
Steps to reproduce the behavior:
1. Select 0.08mm default preset
2. Add a cylinder or shpere primitive
3. Scale object down to 5x5x5
4. Add raft of 1 layer
5. Slice

**Expected behavior**
Expected to slide successfully. Instead, I get either a "bad alloc" error or the program completely crashes.

**Desktop (please complete the following information):**
 - Ubuntu 22.04.2 LTS

tangketan, 2023-05-11:
Thanks for reporting the bug. I'll fix it in the next version.

KiwiDamo, 2023-05-22:
Can confirm this problem also exists in the Windows version.

Laess3r, 2023-06-05:
Also confirm this crash on Windows version. Very annoying, as tiny objects must be print using raft - which is not possible 👎🏻

NicGibbens, 2023-07-10:
Also happens with Mac version.

pedro-pedrosa, 2023-07-10:
apparently a work around is to add normal(manual) supports to the object. this prevents the crash

tangketan, 2023-07-12:
The latest beta should have fixed this issue. Please take a try.

NicGibbens, 2023-07-12:
Now working on Mac using the latest Beta - many thanks!!

closed 2023-07-18 by tangketan