bambulab/BambuStudio #1792: Remove redundant purge line

Lesani, 2023-05-11:
**Is your feature request related to a problem? Please describe.**
I'm furstrated by the existence of two purge lines.
[image]
I think one is used before doing flow calibration, and one after, but I typically turn off flow calibration, and still have both lines.
Especially the top-right one is in my way very often.
Both are printed right after each other, so not really necessary when not using flow calibration

**Describe the solution you'd like**
Move to a single purge line (preferrably the lower one, like prusa)
Modify the lower line to not have a right-angle

**Describe alternatives you've considered**
My alternative on prints where I Can't manage to stack them next to the line neatly, I let it print over this line and need to post-process the part that has the line sticking out of it

SaltWei, 2023-05-24:
@sodachen22 Please ask printer team whether can improve the provided start gcode as user's suggestion

bani6809, 2023-08-01:
purge/prime lines on the bed are completely unnecessary when you can just purge into the poop chute and wipe nozzle. you can also purge into the poop chute a lot faster than extruding lines on the bed.

BrockPlaysFortnite, 2023-10-09:
Is this still not added I don't see why we need that left over purge line from flow calibration when I don't actually use flow calibration

AnhDuck, 2023-12-21:
Yeah the second purge should be removed if you're not using calibration. It's kind of ridiculous for the printer to waste filament (and human time trying to remove it) for absolutely no reason. This change wouldn't even take half a day to implement and be a nice quality of life improvement.

longwa, 2024-01-17:
This issue is even worse than just having two lines. Even if you turn off flow calibration, the first prime line at the front of the plate prints and then the nozzle cools slightly while waiting in the front right corner. After a few seconds, it then heats back up (for some reason) but a little bit of plastic oozes out of the nozzle during this time.

When it prints the rear purge line, that plastic has a good chance of being pushed up against the nozzle and not being removed, only to be deposited somewhere in the print.

It's honestly the most baffling start gcode for a printer that is almost hands off. Just print a normal, single long purge line in the front and then start printing immediately like every other printer if you aren't doing the flow calibration.

sanguinaria-canadensis, 2024-01-17:
will optimize this later.

longwa, 2024-01-17:
So I believe the issue is the M104 S140 on line 437:
```
M1002 judge_flag extrude_cali_flag
M622 J0
    G0 X231 Y1.5 F30000
    G0 X18 E14.3 F{outer_wall_volumetric_speed/(0.3*0.5)     * 60}
M623

M104 S140  ; <-- This should probably be inside the M622 J1 block earlier
```
Only if `extrude_cali_flag` is enabled (earlier M622 J1 block), at the end we should reduce the nozzle temperature to S140 before doing the "handeye calibration" section to prevent oozing.

However, if the flag is **not** enabled, it will be better prevent a short pause by just leaving the nozzle temp alone since we will pretty quickly reset it and have to wait at line 626.

The only thing I'm not sure is there is some laser calibration in between.

hongkongkiwi, 2024-03-11:
I'm also interested in this, the line at the back means I have to remove my build plate to get at it. If it's located in front I can easily scrape it off without removing build plate.

Stickerpants, 2024-03-29:
I would also like to see configurable settings for this.

Lesani, 2024-03-29:
@sodachen22 @SaltWei can you get this TINY change in start GCode out already?

longwa, 2024-06-03:
IMO there are a couple of fixes that are ideal:

1. If the flow calibration is disabled, just print a single purge line ideally across the front since it's easier to remove. That line should be a single pass across with a quick wipe at the end to prevent a string and then immediately into the print. Whatever is happening with that cool down, laser calibration step is a terrible especially for PETG b/c it WILL ooze during that period, guaranteed.

2. If the flow calibration IS enabled, do that in the front and then either go straight into the print (skipping the rear purge) or at least make the rear purge A SINGLE LINE. The fact it makes an L shape round trip doubles down on the issue. Many times, the little blob that formed in step 1 (with the little cooldown/up cycle) is deposited on the back purge line. But the purge goes out and then comes back right next to itself. That will pickup that little blob. It's almost like the printer is trying to ensure there's plastic stuck to the nozzle.

Purge lines should be a single pass, starting out a little high and heavy flow to release oozing to the plate and then moving down slightly with a slight wipe at the end to prevent a long thing string (usually with PETG). There are plenty of good gcode examples out there, just use one.

Pastitas, 2024-06-05:
I'd even go as far as to say that a purge line is NOT NEEDED as is, for example the project https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging uses a purge blob or logo, which cleans the tip every time without issue with less plastic use and a way smaller footprint.

andrewprocter, 2024-07-12:
Being able to move the purge line in the slicer UI rather than in gcode would be nice too

BrockPlaysFortnite, 2024-08-20:
how is this not fixed yet 🗿 I'm too scared to mess with the Gcode for the initial flow calibration purge line and the nozzle temp reset it does the gcode is so complex lol. Removing the back purge line is really easy though just ":" the last lines of gcode but the nozzle heater reset is still causing me big problems with my filament oozing since I don't dry every single roll I have

andrewprocter, 2024-10-09:
@sanguinaria-canadensis @sodachen22 just wondering if there's been any movement on this? I am starting to print 100x of a part where I have to manually remove this line every time 😅