bambulab/BambuStudio #1793: Better exclusion zone handling

Lesani, 2023-05-11:
**Is your feature request related to a problem? Please describe.**
Check if any movements actually collide with the cutting area during slicing, instead of goig for the "outside collision box" of a part
[image]

This is only a visual example, this lizard may actually have travel paths between the body and head that could be an issue. But I have had bigger parts that were just big enough to clip the exclusion zone with their bouding box, while the actual printed are was FAR away from it.

**Describe the solution you'd like**
Change the Error to a Warning
Let the slicer slice, and check in the sliced G-Code if there is an actual bounding box violation
Show an Error if there is an actual violation.

You already do the same with intersecting parts

**Describe alternatives you've considered**
I can turn off the exclusion zone, but this is a hassle doing that in the printer settings every time, when the slicer is so strongly connected to the machine, this should be done automagically

Hologos, 2023-05-13:
If the seem happens to be in the front and then a travel move to right leg happens, it would result in a colision. It's not that easy imho.

Lesani, 2023-05-13:
[image]

For 2 objects that surround the exclusion zone it seems to work already, paths between the objects are only put in a way where they do not intersect the exclusion zone.... may also be accidental though...

[image]

what would happen here? would the printer destroy the filament cutting mechanism? Could such travel paths not be calculated to go AROUND the zone instead, IF they intersect? That could just as well be done if it is a single part ly the salamander above, and if for whatever reason no valid solution for a path is found, THEN error out and say a path intersects the exclusion zone

tangketan, 2023-08-10:
This feature is in our developing list.

benjaminhale, 2024-04-20:
Whenever this is addressed, whether an object has a Brim needs to be considered. Adding the brim makes for strange error behavior if the brim is calculated for a single object to wrap even a little bit around the corner. 

Brim not wrapping around the corner: no error.
[image]

Brim wrapping a tiny bit around the corner: error.
[image]

benjaminhale, 2024-04-20:
Here it gets weirder.... Three objects pretty much horizontal with brims get printed:
[image]

Move the right-most object down by 1mm and it errors.
[image]

BradCOnGitHub, 2025-02-23:
Any updates on this? It greatly reduces the printable area for objects that seek to actually use the entire available bed space