bambulab/BambuStudio #1796: Can't build Windows or OSX (Mac Studio)

traderhut, 2023-05-11:
Windows build gives error 9009 (which I've been told is File not found)  when I try to build it (VS 2019 Professional), Windows 10

Mac Studio / M1 Ultra 128 GB RAM.
Followed instructions (which use a ~ for the home directory instead of /Users/username, that way it will work on everyone's machine, but I used /Users/and put in my username).

Worked right up to the final 'cmake' which said can't find boost.

I tried using the XCode build, and it gave me the same error:

CMake Error at /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR system filesystem thread
  log locale regex chrono atomic date_time iostreams) (Required is at least
  version "1.66.0")
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2377 (find_package_handle_standard_args)
  CMakeLists.txt:343 (find_package)

**3mf File for This Bug**
N/A, not been able to get it to build on any OS

**To Reproduce**
Follow Wiki install instructions

**Expected behavior**
I expected it to build

**Screenshots**
N/A

**Desktop (please complete the following information):**
Windows:  
Edition	Windows 10 Enterprise
Version	21H2
OS build	19044.2728
Experience	Windows Feature Experience Pack 120.2212.4190.0

Processor	Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz   3.70 GHz
Installed RAM	48.0 GB
Product ID	00329-10280-00000-AA157
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

OSX:
 Mac Studio, 2022
 Apple M1 Ultra
Memory: 128 GB
macOS: Ventura 13.3.1
448GB free on disk.

traderhut, 2023-05-11:
I did a 'sudo port install boost'   after installing the port command, and I'm still getting the same error.  I think there is something that is missing in the build steps, something required other than the things installed at the start, I assume that now I have 'boost' (Well, Boost it says) - which seems to be a C++ library that is pretty old (not trying to imply it is bad or anything just because it has been around a while, so have I :-) ), I presume I need to set the Boost_INCLUDE_DIR or something in the build process?  Maybe it is set during some part of the process?   Perhaps starting over??

traderhut, 2023-05-11:
The error message says it can't find it, turns right around and says it found a suitable version, so if that is true, the first part is false, and it COULD find it, the directory it is set to: /opt/local/include has a directory called 'boost', and the things it says it can't find (system, etc) are in that directory... 

CMake Error at /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Boost (missing: system filesystem thread log locale regex
  chrono atomic date_time iostreams) (found suitable version "1.76.0",
  minimum required is "1.66.0")
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2377 (find_package_handle_standard_args)
  CMakeLists.txt:343 (find_package)

So, no clue what is really missing (And I guess I installed the port before checking if it was already in this directory or something)

Is my computer too new?   Need an older OS or something?

traderhut, 2023-05-11:
OK, I found a flag I could add to debug boost, it looks like it finds it fine, but then doesn't bother to include the directory that it found it in in the search path, but has C:/boost/lib in the path (Ah, this is a UNIX machine WTF you got C: in your search for??)

-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1690 ] _boost_TEST_VERSIONS = "1.81.0;1.81;1.80.0;1.80;1.79.0;1.79;1.78.0;1.78;1.77.0;1.77;1.76.0;1.76;1.75.0;1.75;1.74.0;1.74;1.73.0;1.73;1.72.0;1.72;1.71.0;1.71;1.70.0;1.70;1.69.0;1.69;1.68.0;1.68;1.67.0;1.67;1.66.0;1.66"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1691 ] Boost_USE_MULTITHREADED = "TRUE"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1692 ] Boost_USE_STATIC_LIBS = "ON"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1693 ] Boost_USE_STATIC_RUNTIME = <unset>
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1694 ] Boost_ADDITIONAL_VERSIONS = <unset>
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1695 ] Boost_NO_SYSTEM_PATHS = <unset>
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1727 ] BOOST_ROOT = <unset>
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1728 ] ENV{BOOST_ROOT} = <unset>
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1729 ] BOOST_INCLUDEDIR = <unset>
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1730 ] ENV{BOOST_INCLUDEDIR} = <unset>
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1731 ] BOOST_LIBRARYDIR = <unset>
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1732 ] ENV{BOOST_LIBRARYDIR} = <unset>
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1820 ] location of version.hpp: /opt/local/include/boost/version.hpp
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1860 ] Boost_VERSION = "107600"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1861 ] Boost_VERSION_STRING = "1.76.0"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1862 ] Boost_VERSION_MACRO = "107600"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1863 ] Boost_VERSION_MAJOR = "1"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1864 ] Boost_VERSION_MINOR = "76"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1865 ] Boost_VERSION_PATCH = "0"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1866 ] Boost_VERSION_COUNT = "3"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1890 ] Boost_LIB_PREFIX = ""
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1891 ] Boost_NAMESPACE = "boost"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:953 ] _boost_COMPILER = <unset> (guessed)
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:1927 ] _boost_MULTITHREADED = "-mt"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2005 ] _boost_ARCHITECTURE_TAG = "" (detected)
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2009 ] _boost_RELEASE_ABI_TAG = "-"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2010 ] _boost_DEBUG_ABI_TAG = "-d"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2070 ] _boost_LIBRARY_SEARCH_DIRS_RELEASE = "/opt/local/include/lib;/opt/local/include/../lib;/opt/local/include/stage/lib;PATHS;C:/boost/lib;C:/boost;/sw/local/lib"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2071 ] _boost_LIBRARY_SEARCH_DIRS_DEBUG = "/opt/local/include/lib;/opt/local/include/../lib;/opt/local/include/stage/lib;PATHS;C:/boost/lib;C:/boost;/sw/local/lib"
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2256 ] Searching for SYSTEM_LIBRARY_RELEASE: boost_system-mt-1_76;boost_system-mt;boost_system-mt;boost_system-mt;boost_system
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2311 ] Searching for SYSTEM_LIBRARY_DEBUG: boost_system-mt-d-1_76;boost_system-mt-d;boost_system-mt-d;boost_system-mt;boost_system
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2256 ] Searching for FILESYSTEM_LIBRARY_RELEASE: boost_filesystem-mt-1_76;boost_filesystem-mt;boost_filesystem-mt;boost_filesystem-mt;boost_filesystem
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2311 ] Searching for FILESYSTEM_LIBRARY_DEBUG: boost_filesystem-mt-d-1_76;boost_filesystem-mt-d;boost_filesystem-mt-d;boost_filesystem-mt;boost_filesystem
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2256 ] Searching for THREAD_LIBRARY_RELEASE: boost_thread-mt-1_76;boost_thread-mt;boost_thread-mt;boost_thread-mt;boost_thread
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2311 ] Searching for THREAD_LIBRARY_DEBUG: boost_thread-mt-d-1_76;boost_thread-mt-d;boost_thread-mt-d;boost_thread-mt;boost_thread
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2256 ] Searching for LOG_LIBRARY_RELEASE: boost_log-mt-1_76;boost_log-mt;boost_log-mt;boost_log-mt;boost_log
[... cut ...]
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2311 ] Searching for IOSTREAMS_LIBRARY_DEBUG: boost_iostreams-mt-d-1_76;boost_iostreams-mt-d;boost_iostreams-mt-d;boost_iostreams-mt;boost_iostreams
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2256 ] Searching for LOG_SETUP_LIBRARY_RELEASE: boost_log_setup-mt-1_76;boost_log_setup-mt;boost_log_setup-mt;boost_log_setup-mt;boost_log_setup
-- [ /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindBoost.cmake:2311 ] Searching for LOG_SETUP_LIBRARY_DEBUG: boost_log_setup-mt-d-1_76;boost_log_setup-mt-d;boost_log_setup-mt-d;boost_log_setup-mt;boost_log_setup

traderhut, 2023-05-11:
OK, after running (guessing)
brew install boost

I'm getting THIS error:
CMake Error at /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  TBB library cannot be found.  Consider set TBBROOT environment variable.
  (missing: TBB_INCLUDE_DIRS TBB_LIBRARIES)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindTBB.cmake.in:286 (find_package_handle_standard_args)
  cmake/modules/FindTBB.cmake:29 (include)
  CMakeLists.txt:397 (find_package)

Guessing that the instructions didn't include 'brew install tbb'??

traderhut, 2023-05-12:
doing the brew install tbb didn't help (it DID find a 'tbb' package)  I tried removing the CMakeCache.txt and the CMakeFiles directories as the cache file said it couldn't find the TBB_tbb_LIBRARY_[Release/debug], but it WAS able to find the include folder..

I presume that the developers added a bunch of stuff and didn't update the instructions on how to actually build this?

Is there a better place to report this?  I was looking for a forum or something..

Last error message:  (Yeah, confirmed that the /opt/homebrew/include/tbb directory exists with lots of stuff in it, but no, no tbb_stddef.h file - Apparently it got deprecated in 2021...  Oh, joy, which means I either need an old version - there was a suggestion of installing ' brew install tbb@2020' - however, That version is old, and has been disabled because it is not supported upstream... Oh, joy..

Current error message:

CMake Error at cmake/modules/FindTBB.cmake.in:198 (file):
  file failed to open for reading (No such file or directory):

    /opt/homebrew/include/tbb/tbb_stddef.h
Call Stack (most recent call first):
  cmake/modules/FindTBB.cmake:29 (include)
  CMakeLists.txt:397 (find_package)

CMake Error at /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  TBB library cannot be found.  Consider set TBBROOT environment variable.
  (missing: TBB_LIBRARIES) (found version ".")
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.26.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindTBB.cmake.in:286 (find_package_handle_standard_args)
  cmake/modules/FindTBB.cmake:29 (include)
  CMakeLists.txt:397 (find_package)

lanewei120, 2023-05-15:
@traderhut could you show the detailed cmake commands when you building?

did you follow the guide in https://github.com/bambulab/BambuStudio/wiki

traderhut, 2023-05-15:
Yes, I followed those steps exactly, as shown in the Wiki for the Windows
compile.   I gave up on that one as it was getting the error 9009.

I was able to get OrcaSlicer to build on OS-X (Gave up on Bambu Studio, ad
OrcaSlicer is based on that source, so I figured I'd see if I had any
better luck there),
It needed several components that are NOT included in the dependencies in
the Wiki (Either Bambu's or the OrcaSlicer's one, which was basically the
same instructions if I recall), *boost *was one, I think the other one I
listed in the error report (*TBB *had to be installed, but I had to switch
from it being installed using a 'port' (which was someone else's solution -
NOT a good idea), to installing it via Brew, which worked (I just guessed
that maybe it existed there).  That was also not listed as a dependency in
the Wiki.

I think that someone is assuming that you have both of those installed
already if you are using that machine for development purposes... However,
this is about the first thing that I've used the Mac Studio for to develop.

I'd love to be able to modify the slicer to upload the file in a more
intelligent fashion..  I mean, for the video to work, the printer and the
slicer have to be on the same network... so, logically, instead of taking
the Gcode file and uploading it to China and downloading it back, just
connect to the printer and push it there..  Kind of a duh sort of thing..

Anyway, that is off the topic...  I still can't use the printer for
business related purposes until this gets fixed, and verified by monitoring
all traffic going out of the printer in a test setup to make sure that
nothing is being uploaded that should not be (like models to be printed!)
but it makes a fun toy in the meantime!

 -Chert

lanewei120, 2023-05-15:
these missed components are built from depedences
[image]

did you build deps before building the slicer?

lanewei120, 2023-06-09:
@traderhut 
do you still have issues when building?

dbadavekc, 2023-07-02:
I am in the same situation as the OP. I manually installed Boost and TBB, but are getting this when trying to build deps:
youthsonic@Daves-MBP build % cmake ../ -DDESTDIR="/Users/youthsonic/code/BambuStudio/BambuStudio_deps" -DOPENSSL_ARCH="darwin64-arm64-cc"
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: OFF
-- SLIC3R_DESKTOP_INTEGRATION: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: /Users/youthsonic/code/BambuStudio/BambuStudio_deps/usr/local (from cache or command line)
-- PROJECT_SOURCE_DIR: /Users/youthsonic/code/BambuStudio
-- CMAKE_MODULE_PATH: /Users/youthsonic/code/BambuStudio/cmake/modules/
OS X SDK Path: /Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk
OS X Deployment Target: 10.15
-- LIBDIR: /Users/youthsonic/code/BambuStudio/src
-- LIBDIR_BIN: /Users/youthsonic/code/BambuStudio/build/src
-- Boost::boost exists
CMake Error at cmake/modules/FindTBB.cmake.in:198 (file):
  file failed to open for reading (No such file or directory):

    /opt/homebrew/include/tbb/tbb_stddef.h
Call Stack (most recent call first):
  cmake/modules/FindTBB.cmake:29 (include)
  CMakeLists.txt:397 (find_package)

CMake Error at /opt/homebrew/Cellar/cmake/3.26.4/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  TBB library cannot be found.  Consider set TBBROOT environment variable.
  (missing: TBB_LIBRARIES) (found version ".")
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.26.4/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindTBB.cmake.in:286 (find_package_handle_standard_args)
  cmake/modules/FindTBB.cmake:29 (include)
  CMakeLists.txt:397 (find_package)

-- Configuring incomplete, errors occurred!
youthsonic@Daves-MBP build %

dbadavekc, 2023-07-02:
a bit of google-fu seems to point to a requirement for an older tbb version "tbb@2020", which was disabled in brew on "2023-02-07", because: :unsupported... Still digging, but /opt/homebrew/include/tbb/tbb_stddef.h was depricated in more recent tbb versions

lanewei120, 2023-07-03:
@dbadavekc 

have you build the deps firstly before bambu studio?
[image]

dbadavekc, 2023-07-09:
Yes; I had mentioned that I did in my post. Orcaslicer builds perfectly FWIW

lanewei120, 2023-08-08:
boost should be built from the deps, no need to to be installed

I have tried the steps again, I can build it sussccefully on my macos

if still failed, please record all the building logs(including build the deps) into a file and upload here
we can check more

closed 2023-08-18 by lanewei120