bambulab/BambuStudio #1798: P1P Filament Unload Process Having Issues

rashuge, 2023-05-12:
**Describe the bug**
When unloading filament, sometimes the device detects that there is still filament in the extruder.  When in fact there is not.  I have had it with silk PLA.  Effectively it then says "RETRY" for the unload operation when the filament is already out.  It does not have the ability for me to hit "DONE" or anything but "RETRY".  I have to power off the machine and turn it back on to clear the issue.  I will not even let me manually extrude while it is in the "RETRY" state.

**To Reproduce**
Steps to reproduce the behavior:
1. Load SILK PLA
2. Just as the unload says to pull the filament, pull the filament
3. Sometimes it will indicate that the unload failed and press "RETRY".  But there is no filament in the extruder so that will not work.
4. No actual error, just a retry.

**Expected behavior**
I should be able to press "DONE" to indicate that it worked and the filament came out as another option.  The other thing might be that I am pulling the filament a little bit too early but it is free to pull and has been cut.

jlsmith1000, 2023-05-12:
Same issue for first time today, not sure how to clear it.  Power cycle no good.

mechengineermike, 2023-07-19:
Boosting this & your recommended solution. Currently the ONLY option is RETRY. The user needs to be able to cancel & exit that menu.

bolsoncerrado, 2023-07-31:
This is not exclusive to P1P and happens to the X1 series too. 

I too have an extruder with a "clogged" filament sensor so there's always filament present no matter what.

SO

If I use the UNLOAD feature, I have to turn off the printer completely in order to recover control of the machine. The only other solution I've found to avoid this is to start a CALIBRATION TEST when printer is awaiting filament unload and then cancel the whole thing. 

There should be a CANCEL button added when UNLOAD fails for the case expressed above and you don't have phyisical time to replace the extruder or fix the hardware issue...

Can't believe there's no fix for this yet and can't believe no one else has complained on this LARGE user base these printers have!! :O

chasingclouds, 2023-11-08:
Same issues here after 3months on my p1p.
Only fix i did that let me at least load filament manually by using the extruder stepper buttons in the menu is disable the filament runoutsensor in the machine gcode in the slicer :)

bolsoncerrado, 2023-11-08:
I'm still amazed a simple CANCEL button hasn't been added yet to the GUI so you can skip a loading/unloading failure...

vtjballeng, 2024-01-23:
We've now lost multiple 10hr+ prints in expensive PAHT-CF because of this issue. We need a manual override capability when the process fails or the filament sensor is jammed.

vtjballeng, 2024-01-23:
@QingZhangBambu any help here is appreciated.

QingZhangBambu, 2024-01-24:
I will talk to the concerned staff abou that

pichen2021, 2024-01-25:
Hi @rashuge 
Does this issue only occur with PLA-silk? The filament sensor may be abnormal due to debris, suggest to refer to this [wiki ](https://wiki.bambulab.com/en/ams/troubleshooting/ams-loading-unloading-failure) to clean it, cancel this step and continuing to print may cause other abnormalities.
[image]

bolsoncerrado, 2024-01-25:
This is a flaw in the GUI. It can happen with ANY filament.

If for whatever reason ANY filament clogs the sensor your only option is to turn off the machine because the only option in the GUI is to RETRY or Die.

dpmott, 2024-06-08:
Got this today:

[image]

It's been on that screen for over 5 minutes, animating the ellipsis (...), but otherwise doing nothing.

If I remove the tool head cover, Bambu Studio will alert me that the tool head cover has fallen off.  So, it's not as if the software is locked up.  There's just no exit path from this UI condition.  Which is odd, because I usually get the "Retry" option after it attempts to eject the filament, but I'm not there to pull the filament back.

In this case, I'm printing TPU.  I think my cutter is worn and failing to cut the filament completely, so it also pulls up the bit in the heat break.  In this case, I pulled it out by hand, and it looks like a successful cold pull - the tip is pointy.

Okay, it might think that there's something in the filament travel path.  I'd like to be able to preheat the hotend, run the extruder, and feed filament into it to see if I can clear it that way.  But no, I can't even preheat the hotend.  Since it's below temp, it won't move the extruder.  My only hope is that if I power cycle the machine, it'll offer to resume the print.

bolsoncerrado, 2024-06-08:
Sometimes getting into the handy app helps out but I agree the unloading and loading screens deserve a cancel button or more user control rather than endless timeouts...

gaolegao-lx, 2024-06-10:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

bolsoncerrado, 2024-06-11:
This is more than just a ticket but a firmware behaviour where a CANCEL button/option is required during the load/unload process, much like what other czech printers do...