bambulab/BambuStudio #1799: Bambu Studio 1.6.2.4 not launching after update

jimi0303, 2023-05-12:
Hello!

I got a prompt to update the Bambu Studio software on my Windows 11 computer. The update completed successfully, but since then the program won't start. It just hangs on a grey screen with a white top bar, with the windows mouse loading icon circling forever, until I force close it.

Here is what is displays:

[image]

I don't know if it's related, but it also can't display the launch screen properly, most of the bottom text is missing:

[image]

Rebooting the computer didn't do anything. [I've seen one other issue just like this on GitHub already](https://github.com/bambulab/BambuStudio/issues/1770), so I went and renamed the BambuStudio folder in AppData, which also didn't help.

I use this machine for work, so I'd really like to get it working again. Any help is appreciated.

bambu123, 2023-05-15:
Please send us the logs from “%APPDATA%/BambuStudio".

Whenever it is locking, please use the Windows Task Manager to create the minidump of BambuStudio, the dump is useful to analyze the hung problem and please send it to us.

bambu123, 2023-05-15:
You can create minidump with the following steps:

Open the Windows Task Manager
Find and select the Bambu Studio process in it
Right click the process and choose the 'Create Minidump File' menu item
Select a folder to save the file

bambu123, 2023-05-19:
@jimi0303 Are you still facing this issue? Please help make a minidump if yes.

Bukaro21, 2023-06-19:
@bambu123 I have the exact same issue with bambu studio opening but staying white and getting a "BambuStudio is not responding" message from Windows until I force close it. I created a minidump and uploaded it to WeTransfer as it was too big to attach here: https://we.tl/t-uU9fRJ3KDe

bambu123, 2023-06-19:
> @bambu123 I have the exact same issue with bambu studio opening but staying white and getting a "BambuStudio is not responding" message from Windows until I force close it. I created a minidump and uploaded it to WeTransfer as it was too big to attach here: https://we.tl/t-uU9fRJ3KDe

Great thanks. I will check it.

bambu123, 2023-06-19:
> @bambu123 I have the exact same issue with bambu studio opening but staying white and getting a "BambuStudio is not responding" message from Windows until I force close it. I created a minidump and uploaded it to WeTransfer as it was too big to attach here: https://we.tl/t-uU9fRJ3KDe

Are you openning a 3mf file or starting Bambu Studio directly without 3mf file?
Please provide the app logs to us, which can be found from the menu "Help > Open Configuration Folder".

Bukaro21, 2023-06-19:
[image]
[image]
[image]

[log.zip](https://github.com/bambulab/BambuStudio/files/11787046/log.zip)

Here are screenshots and a zip of all logfiles I have. I am trying to open Bambu Studio directly without a 3mf file

Bukaro21, 2023-06-20:
@bambu123 It seems to have to do with multi monitors. Once I disabled my other monitors Bambu Studio started fine

Braindea7, 2023-06-20:
> @bambu123 It seems to have to do with multi monitors. Once I disabled my other monitors Bambu Studio started fine

I can confirm this. It seems like the problem arises when the program crashes or the internet connection gets lost on a display other than the main display.

Disconnected the non main displays, start the studio and reconnect the displays. Everything fine again.

Gili-Omri, 2023-06-21:
@bambu21 

Good call

Same issue, on Windows 10, worked fine yesterday, problem started today
disconnecting all displays but main , solves the issue

any idea how will the fix take?

bambu123, 2023-07-12:
> Whenever it is locking, please use the Windows Task Manager to create the minidump of BambuStudio, the dump is useful to analyze the hung problem and please send it to us.

Whenever it is locking, please use the Windows Task Manager to create the minidump of BambuStudio, the dump is useful to analyze the hung problem and please send it to us.

You can create minidump with the following steps:

Open the Windows Task Manager
Find and select the Bambu Studio process in it
Right click the process and choose the 'Create Minidump File' menu item
Select a folder to save the file

bambu123, 2023-07-12:
> @bambu21
> 
> Good call
> 
> Same issue, on Windows 10, worked fine yesterday, problem started today disconnecting all displays but main , solves the issue
> 
> any idea how will the fix take?

Your two screens may have different scale ratios. We found it relevant to scale change.
We are trying to fix it. Setting the same scale ratio may avoid this issue currently.

bambu123, 2023-08-28:
@jimi0303
The new version 1.7 was released. 
Please help to see if this problem is solved.

jimi0303, 2023-08-29:
Hello @bambu123 !

Thank you for the follow-up. After my initial report I deleted everything related to Bambu Studio on my computer and reinstalled it from scratch. This solved the issue for me and the problem hasn't resurfaced since.

cmguo, 2023-08-29:
@Gili-Omri @Braindea7 @Bukaro21 
Please help to see if this problem is solved in the new version 1.7.

bambu123, 2023-08-29:
@Gili-Omri @Braindea7 @Bukaro21
Please help to see if this problem is solved in the new version 1.7.

Alamin-sajib, 2023-09-02:
> @Gili-Omri @Braindea7 @Bukaro21 Please help to see if this problem is solved in the new version 1.7.

Same issue in latest v1.7 release.

bambu123, 2023-09-04:
> > @Gili-Omri @Braindea7 @Bukaro21 Please help to see if this problem is solved in the new version 1.7.
> 
> Same issue in latest v1.7 release.

Please provide all the logs.

mverch67, 2023-09-29:
This issue happens also in version 1.7.6.

When the window size on the second monitor is bigger than on the main monitor is will crash, e.g. when doing a maximize window or resizing and exceeding resolution of the main monitor.

DanBao-Bambu, 2024-01-16:
@mverch67 @Alamin-sajib 
I need your log and minidump files. Additionally, I can try to see if v1.8 has resolved this issue. The v1.8 log and minidump files are user-friendly for my debugging.

>You can create minidump with the following steps:

>Open the Windows Task Manager
Find and select the Bambu Studio process in it
Right click the process and choose the 'Create Minidump File' menu item
Select a folder to save the file

Thank you very much

mverch67, 2024-01-16:
> @mverch67 @Alamin-sajib I need your log and minidump files. Additionally, I can try to see if v1.8 has resolved this issue. The v1.8 log and minidump files are user-friendly for my debugging. [...]

Yes, this issue is resolved in v1.8, no more problems, thanks!

closed 2024-02-06 by bambu123