bambulab/BambuStudio #1800: Only switch loaded AMS filament if necessary

akudaikon, 2023-05-12:
**Is your feature request related to a problem? Please describe.**
After every print, the AMS unloads the filament from the printer. If the next print uses the same AMS slot, then it has to reload and purge the filament again. If you're doing multiple prints from the same AMS slot, this is time consuming and a waste of filament.

**Describe the solution you'd like**
Don't unload the filament at the end of every print. If the next print uses the same AMS slot, then it is already loaded from the print before. If it does not use the same AMS slot, then unload the filament and change.

Hologos, 2023-05-13:
Alter your end gcode to have this behaviour.

psjshep, 2023-05-13:
suggestions on how to do this with an example, perhaps, would be very useful...

mobomelter, 2023-05-14:
This would be a pretty awesome feature to have to save on time/purged filament.